xdk2mam/xdk2mam #5: Compiling error xdk2mam-http

Deca, 2019-01-08:
Hi, I'm trying to compile the xdk2mam-http but end up with errors, is the actual C code on the repository tested and working? 
I edited the parameters on  xdk2mam.h did a project clean & build, those are the errors I'm getting:

`Creating .out C:/Users/Decagrog/Documents/XDK-Workbench/xdk2mam-master/http/xdk2mam-c/debug/xdk2mam-html.out
./../Libraries/ServalStack/debug/libServalStack_efm32_debug.a(HttpClient.o): In function `openTcpConnection':
C:\Users\Decagrog\Documents\XDK-Workbench\XDK\SDK\xdk110\Libraries\ServalStack/3rd-party/ServalStack/src/Http/HttpClient.c:236: undefined reference to `Tcp_connectSecure'
./../Libraries/ServalStack/debug/libServalStack_efm32_debug.a(HttpClient.o): In function `doReceive':
C:\Users\Decagrog\Documents\XDK-Workbench\XDK\SDK\xdk110\Libraries\ServalStack/3rd-party/ServalStack/src/Http/HttpClient.c:541: undefined reference to `Tls_getSocketStatus'
C:\Users\Decagrog\Documents\XDK-Workbench\XDK\SDK\xdk110\Libraries\ServalStack/3rd-party/ServalStack/src/Http/HttpClient.c:552: undefined reference to `Tls_receive'
./../Libraries/ServalStack/debug/libServalStack_efm32_debug.a(HttpPool.o): In function `HttpPool_close':
C:\Users\Decagrog\Documents\XDK-Workbench\XDK\SDK\xdk110\Libraries\ServalStack/3rd-party/ServalStack/src/Http/HttpPool.c:231: undefined reference to `Tls_close'
./../Libraries/ServalStack/debug/libServalStack_efm32_debug.a(HttpPool.o): In function `HttpPool_delete':
C:\Users\Decagrog\Documents\XDK-Workbench\XDK\SDK\xdk110\Libraries\ServalStack/3rd-party/ServalStack/src/Http/HttpPool.c:299: undefined reference to `Tls_delete'
./../Libraries/ServalStack/debug/libServalStack_efm32_debug.a(HttpPool.o): In function `HttpPool_subMonitorFunc':
C:\Users\Decagrog\Documents\XDK-Workbench\XDK\SDK\xdk110\Libraries\ServalStack/3rd-party/ServalStack/src/Http/HttpPool.c:451: undefined reference to `Tls_close'
C:\Users\Decagrog\Documents\XDK-Workbench\XDK\SDK\xdk110\Libraries\ServalStack/3rd-party/ServalStack/src/Http/HttpPool.c:477: undefined reference to `Tls_delete'
./../Libraries/ServalStack/debug/libServalStack_efm32_debug.a(TcpMsg.o): In function `sendViaTcp':
C:\Users\Decagrog\Documents\XDK-Workbench\XDK\SDK\xdk110\Libraries\ServalStack/3rd-party/ServalStack/src/Msg/TcpMsg.c:299: undefined reference to `Tls_prepareForSending'
C:\Users\Decagrog\Documents\XDK-Workbench\XDK\SDK\xdk110\Libraries\ServalStack/3rd-party/ServalStack/src/Msg/TcpMsg.c:337: undefined reference to `Tls_send'
C:\Users\Decagrog\Documents\XDK-Workbench\XDK\SDK\xdk110\Libraries\ServalStack/3rd-party/ServalStack/src/Msg/TcpMsg.c:350: undefined reference to `Tls_retrySendingLater'
collect2.exe: error: ld returned 1 exit status
application.mk:372: recipe for target 'C:/Users/Decagrog/Documents/XDK-Workbench/xdk2mam-master/http/xdk2mam-c/debug/xdk2mam-html.out' failed
mingw32-make[1]: *** [C:/Users/Decagrog/Documents/XDK-Workbench/xdk2mam-master/http/xdk2mam-c/debug/xdk2mam-html.out] Error 1
mingw32-make[1]: Leaving directory 'C:/Users/Decagrog/Documents/XDK-Workbench/XDK/SDK/xdk110/Common'
Makefile:58: recipe for target 'debug' failed
mingw32-make: *** [debug] Error 2

Mind I'm a webdev and never touched a line of C, can read the code more or less but I'm not able to  grasp what is going wrong so it could be also something on my side

demichele, 2019-01-08:
Hi there. Could you let us know what XDK Workbench versión are you using? Right now we recommend to compile our code with v 3.4 since latest 3.5 XDK Workbench release does not compile our code well. This is an issue we are reviewing with the Bosch XDK team.

Deca, 2019-01-08:
Ah! Indeed I was using the v3.5, installed now the 3.4 and it compiled without issue, thanks! I'll keep an eye for the future release

closed 2019-01-09 by demichele

MuthuMaldini, 2019-01-10:
On a side note, there was an important change between v3.4.0 and v3.5.0 of the XDK workbench with respect to security.

Until v3.4.0 of the workbench, ESCRYPT CycurTLS was used as the Security Library.
With v3.5.0 of the workbench, it has been migrated to use the arm mbed TLS.

The network stack used in XDK for Communication is the ServalStack Library.
When used with ESCRYPT CycurTLS, there are a set of macros for ServalStack that managed the Security Library linkage and the ServalStack security mechanism (like SERVAL_TLS_CYCURTLS, EscTls_CIPHER_SUITE and such).

With the introduction of the arm mbed TLS, it gives more freedom to the user to control and manage their security needs since the arm mbed TLS comes with Apache License version 2.0.

This mean that the ServalStack configuration will need some adaptation in the application as per its need for v3.5.0 for using it with arm mbed TLS.
Below are a set of default for configuration that would be needed in the application makefile (refer the HttpExampleClient configuration).

export SERVAL_TLS_MBEDTLS=1
export SERVAL_ENABLE_TLS_CLIENT=1
export SERVAL_ENABLE_TLS_ECC=1
export SERVAL_ENABLE_TLS_PSK=0
export SERVAL_MAX_NUM_MESSAGES=8
export SERVAL_MAX_SIZE_APP_PACKET=900
export SERVAL_ENABLE_TLS=1
export XDK_MBEDTLS_PARSE_INFO=0

In addition, the complete arm mbed TLS library can be further customized via the SDK/Common/config/MbedTLS/MbedtlsConfig.h file which comes with a default configuration for most of the servers.

P.S. - 

If Security is not off interest, I believe enabling SERVAL_ENABLE_TLS and SERVAL_TLS_MBEDTLS SERVAL_TLS_MBEDTLS macros in your application makefile should fix it for v3.5.0 of the Workbench since the secure API's necessary for Tcp_connectSecure, Tls_close and such is available in the adapter implemented in SDK/Common/source/Private/ServalStack/src/TLS_MbedTLS/MbedTLS_Tls.c file.

On a less technical note, just ensuring that the compiler macros enabled in HttpExampleClient/Makefile file is also enabled in your application will fix this.

MuthuMaldini, 2019-01-10:
@demichele You are right. I see your point here.

XDK2MAM without a doubt is of great interest to Bosch XDK.
The work done in XDK2MAM is pretty impressive especially considering the quick amount of time spent on it and its ability & potential to provide solution to the current market needs.

Providing backward compatibility is definitely not a new topic in software development. I am sure that going forward, XDK2MAM would be one of the projects that the Bosch guys at WB compile before their next releases.

gadavidd, 2020-12-12:
Hey, did we get any solution for this problem?