xdk2mam/xdk2mam #6: Error running node xdk2mam-ble.js scripts on raspberry pi zero

Deca, 2019-01-16:
I'm trying to run the bluetooth-sdcard scripts on a raspberry pi zero WH but shortly after launched it return this error:

`**********************************************************************

{"xdk2mam":[{"sensor": "Environmental","data":[{"Pressure":"101052"},{"Temperature":"32410"},{"Humidity":"29"}]},{"sensor":"Accelerometer","data":[{"x":"-3"},{"y":"29"},{"z":"1008"}]},{"sensor":"Gyroscope","data":[{"x":"-366"},{"y":"-488"},{"z":"-3051"}]},{"sensor":"Inertial","data":[{"x":"-366"},{"y":"-854"},{"z":"0"}]},{"sensor":"Light","data":[{"milliLux":"50040"}]},{"sensor":"Magnetometer","data":[{"x":"32"},{"y":"36"},{"z":"9"}]} ],"device": "248CE70556960C54","timestamp": "1547643215"}
***********************************************************************

thread '<unnamed>' panicked at 'destination and source slices have different lengths', libcore/slice/mod.rs:695:9
note: Run with `RUST_BACKTRACE=1` for a backtrace.
`

The pi zero have a freshly installed raspbian lite,  node v8.9.0, npm 6.5.0
Other scripts (http,mqtt) don't throw errors,  so I can only suppose is something related harwdare differences (maybe the bluetooth?) looking at this[ comparison table](http://socialcompare.com/en/comparison/raspberrypi-models-comparison) is see the pi3 have blue 4.2/ble and pizero blue 4.1

The same bluetooth script is running fine on a raspberry pi3 B+,

demichele, 2019-01-16:
Hi Deca. Please try using a different full node and let us know if the issue persist. Thanks!

Deca, 2019-01-18:
Thanks for the reply Daniel, 

Yes the error persist using different nodes:  I've tried fews from the list https://iota.dance/ (tried combination with IRI 1.5.6/1.6.0, ports 14265/443, pow enabled/disabled), as long my personal node (IRI 1.6.0, pow enabled) 

I don't rule out it could be still something on my side, let me know if there is something else I can do

On a personal note I think that it would be nice having the node scripts running on smaller and less energy hungry devices like the Pi Zero, it's almost the same form factor of the xdk :)