xdk2mam/xdk2mam #8: Error on mqtt-sdcard DEV

Deca, 2019-03-18:
Hi again,

I'm trying the mqtt-sdcard in the DEV repository
Installed workbench 3.5, cleaned, compiled and flashed the mqtt-sdcard. Config.cfg parameters are set
After flashing I get this error:
INFO | XDK DEVICE 1: System Error:, Severity:2 Error code: 67 Package Id:10 and module ID is :2 

Do you have any suggestion?

aleelus, 2019-03-18:
Hi @Deca,
Did you try to turn on the xdk with the micro sd (which contains the file 'config.cfg') inside?

Deca, 2019-03-18:
Yes, I'm running the XDK with a 2gb sdcard and the config.cfg parameters edited for my network

On the xdk forum there is another user that have the same error:
https://xdk.bosch-connectivity.com/community/-/message_boards/message/352830

They suggest him that the package id:10 is related to the BSP package. I did a search for "PACKAGE_ID = 10" in the SDK folder but I can't find anything

aleelus, 2019-03-18:
Hi @Deca,
In the configuration file there should be no space or tabs (unless it is part of a password, name of ssid, etc).

PS: Just in case, the name of the file must be 'config.cfg' without the capital letter C.

Deca, 2019-03-18:
Yes I confirm that there aren't space or tabs of any kind and the file name is lowercase, here a screenshot of the .cfg and the error
[image]

aleelus, 2019-03-18:
@Deca, can you try to put a '\n' or hit enter after the 'YES' of the accoustic sensor (this version of parse seeks the '\n')?

Deca, 2019-03-18:
Added the newline char after the "YES" unfortunately I get the same error. I tried also with another sdcard (sandisk 16gb) just in case

Deca, 2019-03-18:
I finally managed to get a working configuration. 
I had to open the .cfg file with an editor able to show hidden characters and put a carriage return (0x0D) before the line feed to each line, this is beyond to my understanding since I barely have a basic C language background and know that working with buffer and binary mode could be tricky

[image]

aleelus, 2019-03-18:
@Deca,
Thanks for notice this issue, in fact text files created on DOS/Windows machines have different line endings than files created on Unix/Linux. DOS uses carriage return and line feed ("\r\n") as a line ending, which Unix uses just line feed ("\n").
You can change the format in Notepad++ if you go to Edit->EOL Conversion->Windows (CR LF)