xdk2mam/xdk2mam #9: xdk2mam not starting app in XDK

mat-l, 2019-04-03:
Dear all,
we just bought a new Bosch XDK and downloaded XDK2MAM. As described, we changed the wifi settings and compiling does not give any errors. After flashing to the XDK we get the following error or info message:

`INFO | XDK DEVICE 1: 
 INFO | XDK DEVICE 1: XMODEM Download Success
 INFO | XDK DEVICE 1:  CRC of application area
 INFO | XDK DEVICE 1:  CRC0000A5B2
 INFO | Application checksum 'a5b2' successfully verified.
 INFO | Transmission successfully completed!
 INFO | Booting application...
 INFO | XDK DEVICE 1: b
 INFO | XDK DEVICE 1:  Jumping to application
 INFO | Port 'COM4' has been disconnected
 INFO | Connecting to XDK device 'XDK Device 1' in port 'COM4'...
 INFO | Connection to port 'COM4' established
 INFO | XDK DEVICE 1: Connect Failed, so example is not started
 INFO | XDK DEVICE 1: System Error:, Severity:2 Error code: 4 Package Id:30 and module ID is :26 
 INFO | XDK DEVICE 1:  Reset the system`

Can you help us identifying the problem, that we can stream the sensor data via MQTT to our server and connect the XDK to IOTAs Tangle?

Best regards

demichele, 2019-04-03:
Hello @mat-l. Could  you please specify which XDK2MAM MQTT version are you using (MQTT+SDCard or MQTT)? Also, be sure to use the code at the branch that matches your XDK Workbench version. Please, let us know which MQTT pckg you are using. If you are using the SD card triple check to have the config.cfg file with correct values and also to have the card inserted on the XDK110 micro SD slot before you flash the code.

mat-l, 2019-04-03:
Hello @demichele 
We are using XDK110 without SD card in combination with Workbench 3.6 and also the code related to 3.6 which should be the latest one.

aleelus, 2019-04-03:
Hi @mat-l,
That was an old version of the mqtt method, in a few minutes we will upload the updated version of the mqtt method without the sd-scard.
Thanks!

demichele, 2019-04-03:
Hi @mat-l if you get it to work please let us know so we can close this issue. Thanks for the feedback!

mat-l, 2019-04-04:
Perfect, great @demichele @aleelus the updated version is working for us. Thanks for the fast feedback and update. :)

closed 2019-04-04 by mat-l

smileheartravi, 2019-07-20:
can I have XDK 110  code for temperature measurement reading over SD card