xdk2mam/xdk2mam #10: Enabling Acoustic Sensor caused Hardware Reset

chiupc, 2019-04-08:
Hi,

After enabling acoustic sensor together with the other sensors, the device keep resetting it after throwing the following message.
It does not have this error when acoustic sensor alone is enabled.

` INFO | XDK DEVICE 1: WLANPersonalWPA2Connect : Connected to WPA network successfully. 
 INFO | XDK DEVICE 1: NTP request Sending Complete
 INFO | XDK DEVICE 1: Environmental Sensor initialization Success
 INFO | XDK DEVICE 1: Accelerometer Sensor initialization Success
 INFO | XDK DEVICE 1: Gyroscope Sensor initialization Success
 INFO | XDK DEVICE 1: Inertial Sensor initialization Success
 INFO | XDK DEVICE 1: Light Sensor initialization Success
 INFO | XDK DEVICE 1: Magnetometer Sensor initialization Success
 INFO | XDK DEVICE 1: Acoustic Sensor initialization Success
 INFO | XDK DEVICE 1: Broker address: mqtt://172.25.244.91:1883
 INFO | XDK DEVICE 1: EventHandler Event : 0
 INFO | XDK DEVICE 1: Connection Event:
 INFO | XDK DEVICE 1: 	Server Return Code: 0 (0 for success)
 INFO | XDK DEVICE 1: 	Reused Session Flag: 0
 INFO | XDK DEVICE 1: Subscribing to topic: xdk, Qos: 0
 INFO | XDK DEVICE 1: EventHandler Event : 8
 INFO | XDK DEVICE 1: MQTT_SUBSCRIPTION_ACKNOWLEDGED
 INFO | XDK DEVICE 1: HardFault_Occured_going to reset the processor `

demichele, 2019-04-08:
Hey @chiupc we are reviewing this and will come back to you as soon as we have it fixed. Thanks for you feedback!

aleelus, 2019-04-09:
Hi @chiupc , the fix to enable all sensors at the same time is now ready. 
Cheers!

aleelus, 2019-04-09:
Hello  @chiupc,
Yes, of course! we are working on it.