xdk2mam/xdk2mam #11: Invalid application

edgarmartinezg8, 2019-05-29:
Hi Guys, 
I'm trying to test your code for a couple of days, for some reason I can not get it to work. I've tried the following Workbench versions 3.6 & 3.4
In version 3.4 I can not even build the code
In version 3.6 all goes well until I try to flash my device,and the console sends me this message:

INFO | Connecting to XDK device 'XDK Device 1' in port 'COM30' ...
 INFO | Connection to port 'COM30' established
 INFO | Flashing file 'C: /tutorial/xdk2mam/http-sdcard/xdk2mam-c/debug/xdk2mam.bin' ...
 INFO | XDK DEVICE 1: Ready
 INFO | XDK DEVICE 1: C
 INFO | XDK DEVICE 1:
 INFO | XDK DEVICE 1: XMODEM Download Success
 INFO | XDK DEVICE 1: CRC of application area
 INFO | XDK DEVICE 1: CRC000079CA
 INFO | Application checksum '79ca' successfully verified.
 INFO | Transmission successfully completed!
 INFO | Booting application ...
 INFO | XDK DEVICE 1: b
 INFO | XDK DEVICE 1: Invalid application
facts:
SD card.

I know that my card is fine because with it I have tested the connectivity demo where the cronfig.txt need to be safe on the SD card

My PC
I've tried the code on two different computers, both Windows 10 installed 
i just get the same message: Invalid application

Have you any idea whats its going wrong?

Thanks!

demichele, 2019-05-29:
Hi @edgarmartinezg8 
We had this issue from time to time and it seems to be related to something not working fine on the XDK Workbench. We will report this to Bosch XDK110. Meanwhile, what we usually do to get the app flashed and running when we have this issue is: 

1. Use Workbench 3.6 and that branch code. 
2. Turn on your XDK110 on Bootloader Mode (hold the one dot button and turn the XDK110 on, release the button once is one)
3. Right click the project and build it (skip the Clean Project)
4. Click en the Boot button at the XDK Workbench (underneath Flash)
5. Click on Flash

This combination of steps usually works for us. We do not know the reason why this happens though. 
Keep us posted on the outcome!

edgarmartinezg8, 2019-05-31:
Thanks a lot.
it works now ( I tested just Examples http and UDP). I followed  exactly steps 1-5 as you describe.
Just for the Record:
It can be a bit confusing if you have followed the instructions on this Video tutorial  https://www.youtube.com/watch?v=isrx7ibcRL4

For the other users and testers, just made as @demichel says.
Best regards
Edgar

demichele, 2019-05-31:
@edgarmartinezg8 this is an issue that should not occur normally and that's why we did not explained the procedure I gave you above. The code builds well but there seems to be some kind of problem during the flash that goes beyond our code. Already sent an email to the XDK110 team to see if they can find out why this is happening.

closed 2019-05-31 by demichele

burnumburnum, 2019-06-12:
Hello everyone,

I am currently trying to use your code using workbench 3.6. I tried both "ble-sdcard" and "udp-sdcard" trying to follow both instructions from this issue and the official readme. I tried all on MAC and Windows 10. All without success. 

Building works (Without cleaning, the first building process fails, the second works; with cleaning: first building works). When flashing, I also get the info
 INFO | XDK DEVICE 1:  Invalid application 
analogously to edgarmartinezg8

Did you follow any further steps to flash the code? For example, I have the feeling that your procedure only works if a valid application was previously flashed (e.g. Virtual XDK). Maybe there are any further tricks or already code updates or feedback from BOSCH. I'm also not sure when it is best to insert the SD card. Before or after flashing? XDK on or off?

BTW: I just realized that in the config.cfg of the udp package, it says "WLAN_SS**DI**". I assume it should be "WLAN_SSID", instead. 

I would really appreciate a response. THANKS!

edgarmartinezg8, 2019-06-14:
H iburnumburnum_
#Did you follow any further steps to flash the code?__ 

Yes,  but, you need to follow exactly this instrucctions:

1. Use Workbench 3.6 and that branch code.
2. Turn on your XDK110 on Bootloader Mode (hold the one dot button and turn the XDK110 on, release the button once is one) 
---if you do that step right the **Boot** button its gonna be active...--

3. Right click the project and build it (skip the Clean Project)
4 **Click on the Boot button at the XDK Workbench (underneath Flash)**
--- if you click the Boot buttton it will switch to Reboot --
5 Click on Flash

_# I'm also not sure when it is best to insert the SD card. Before or after flashing? XDK on or off?_
I have inserted the SD card before  i turn the device on, 

#BTW: I just realized that in the config.cfg of the udp package, it says "WLAN_SSDI". I assume it should be "WLAN_SSID", instead.

--I assume that to,
 You can use Windows text editor,  Sublime, Geany
To check if you have typed text characters that should not be there

I hope that can help to get it work
Best regards

demichele, 2019-06-14:
Hi @burnumburnum. First of all, the steps provided above are for a XDK110 that has already code flashed (that's why you start it on Bootloader mode by pushing the one dot button). You always need to clean the project and then build. Once this is done, the MicroSD should be inserted before you hit the flash button. So you are expected to, import, clean, build and flash. If this gives you invalid application, there you should try the steps above. The WLAN_SSDI is an error we will correct in the next release, leave it as it is since if you change that the parser wont get the WiFi connection name. We will keep you posted on this issue. We think this might be fixed on next Workbench release. Right now, when we have this issue we do those 5 steps. Sometimes you need to do them a couple of times. It's a pain and we are trying to find out why this started happening with the latest Workbench release.

burnumburnum, 2019-06-16:
Hi everyone,
thank you for your help. I now found a way to flash the code, which slightly differs from your ways:
1. Make sure, a working code is flashed on the XDK. If not, build and flash one of the Demos (I used the VirtualXDKDemo)
2. Put the SD card with the config in the XDK
3. Clean project two times
4. Build project
5. Flash project -> invalid application appears
6. Flash project again -> application works